bhagman/SoftPWM #1: Multiple Pins from one call

mangus580, 2015-09-15:
I would like to activate multiple pins at once using soft pwm...

maybe the ability to pass an array instead of a single value?

Would this be possible?

bhagman, 2015-09-15:
By "activate", do you mean through SoftPWMBegin() or SoftPWMSet()?

That is, do you want to initialize multiple pins at once, or set the values for multiple pins at once?

mangus580, 2015-09-15:
set values for multiple pins at once

mangus580, 2015-09-15:
Maybe I should elaborate...

I have 8 LED light modules.  I want to be able to activate say lights 2,4,6,8 all at the same time.

Problem is, sometimes I want to activate 1,2,3,4 at the same time.  

So, before I decided I was going to use PWM, I was going to set them using PORTD to assign the on/off values to my pins.

Problem is, I really would like to be able to dim my lights.  So I need to be able to use PWM....

bhagman, 2015-09-15:
So wouldn't this work for what you want to do?

```arduino
uint8_t ledSet1[4] = {2, 4, 6, 8};
uint8_t ledSet2[4] = {1, 2, 3, 4};

.
.
.

  int newValue = 55;
  int newValue2 = 99;

  for (int i = 0; i < 4; i++)
    SoftPWMSet(ledSet1[i], newValue);

  for (int i = 0; i < 4; i++)
    SoftPWMSet(ledSet2[i], newValue2);
```

Maybe a code example from you might help.

mangus580, 2015-09-15:
here is an example of a function I am using to strobe the light

int quad_flash(int x){

  while (loop_count < 8) {
  
    SoftPWMSetPercent(x, 100);   // turn the LED on (HIGH is the voltage level)
    delay(35);               // wait for a second
    SoftPWMSetPercent(x, 0);    // turn the LED off by making the voltage LOW
    delay(65);               // wait for a second
    loop_count ++;
  }
  delay (0);
  loop_count = 0;
  
}

Ideally I would be able to pass several lights at once making them flash at the same time

bhagman, 2015-09-15:
How about this?

```arduino
uint8_t ledSet1[4] = {2, 4, 6, 8};
uint8_t ledSet2[4] = {1, 2, 3, 4};

void quad_flash(const uint8_t ledSet[], int ledCount)
{
  for (int flashCount = 0; flashCount < 8; flashCount++)
  {
  	for (int led = 0; led < ledCount; led++)
  	{
      SoftPWMSetPercent(ledSet[led], 100);
  	}

  	delay(35);

  	for (int led = 0; led < ledCount; led++)
  	{
      SoftPWMSetPercent(ledSet[led], 0);
  	}
  	
  	delay(65);
  }
}

.
.
.

  quad_flash(ledSet1, 4);
  quad_flash(ledSet2, 4);
```

mangus580, 2015-09-15:
Works Perfect!!!

Thanks!!

bhagman, 2015-09-15:
:v:

closed 2015-09-15 by bhagman

mangus580, 2015-09-15:
Looking at it now, the lights might stagger flash some...  and not flash together.  I wont know for sure until I have more than 2 lights going (waiting for more parts to arrive) 

I can update later this week...

Thanks
Mike

mangus580, 2015-09-17:
Seems it was the transistors i was using.  I think I am all set now.

Thanks again!