bhagman/SoftPWM #2: millis() stops working when using more than 5 LEDs with SoftPWM

ld21, 2016-05-26:
Hi Brett
I like your SoftPWM-library for arduino a lot. Just today I found out, that millis() stops working, as soon as more than 5 LEDs are controlled using our library. I wasn't able to find a solution to that problem. Do you have an idea?
Best regards,
Dani

I used your example-code (), which uses 8 LEDs:
When checking with the serial monitor, I get constantly
<output>
26
26
26
</output>

<code>
#include <SoftPWM.h>

#define DELAY 100
uint8_t leds[8] = {11,12,13,18,17,16,15,10};

void setup()
{
  Serial.begin(57600);
  SoftPWMBegin();
  
  for (int i = 0; i < 8; i++)
    SoftPWMSet(leds[i], 0);

  SoftPWMSetFadeTime(ALL, 50, 400);
}

void loop()
{
  int i;
  Serial.println(millis());
  for (i = 0; i < 7; i++)
  {
    SoftPWMSet(leds[i+1], 255);
    SoftPWMSet(leds[i], 0);
    delay(DELAY);
  }
  
  delay(400);
  
  for (i = 7; i > 0; i--)
  {
    SoftPWMSet(leds[i-1], 255);
    SoftPWMSet(leds[i], 0);
    delay(DELAY);
  }
  delay(400);
}
</code>
___________________________________________________________________
As soon as I change the code, so that it uses only 5 LEDs, it works as expected:
<output>
3014
5851
8695
</output>

<code>
#include <SoftPWM.h>

#define DELAY 100
uint8_t leds[8] = {11,12,13,18,17,16,15,10};

void setup()
{
  Serial.begin(57600);
  SoftPWMBegin();
  
  for (int i = 0; i < 4; i++)
    SoftPWMSet(leds[i], 0);

  SoftPWMSetFadeTime(ALL, 50, 400);
}

void loop()
{
  int i;
  Serial.println(millis());
  for (i = 0; i < 3; i++)
  {
    SoftPWMSet(leds[i+1], 255);
    SoftPWMSet(leds[i], 0);
    delay(DELAY);
  }
  
  delay(400);
  
  for (i = 3; i > 0; i--)
  {
    SoftPWMSet(leds[i-1], 255);
    SoftPWMSet(leds[i], 0);
    delay(DELAY);
  }
  delay(400);
}
</code>

bhagman, 2016-06-18:
I tested the code that you said didn't work on an Arduino UNO Rev3, using Arduino IDE (SRL) 1.7.10 and Arduino IDE (LLC) 1.6.9.  Works fine.

Please tell us what board you're using and what software you're using.

ld21, 2016-06-20:
Hi Brett, thanks for looking into this. I'm using a LilyPad with a ATMEGA 328V processor (https://www.arduino.cc/en/Main/ArduinoBoardLilyPad). I'm using Arduino IDE 1.6.9 (arduino.cc). Best regards, Dani

bhagman, 2016-06-20:
What voltage are you running the Lilypad at, and how are you powering it?

It sounds like it may be a problem with the power supply.

ld21, 2016-06-20:
I'm either powering it over a 3.3V-FTDI-Cable (then I can monitor the output as posted above) or using 5V from a regulator (but then I don't dare plugging the FTDI-Cable to monitor the thing). But the behavior seems to be the same. The specs say that it is possible to run the Lilypad with voltages from 2.7 to 5.5V and that it runs at 8 MHz (https://www.arduino.cc/en/Main/ArduinoBoardLilyPad).

bhagman, 2016-06-21:
A 3.3V FTDI cable can only supply up to 50 mA @ 3.3V to the board.

If you have selected your current limiting resistors for the LEDs to allow anything near 10mA to each LED, then when all 5 LEDs are on, the voltage regulator on the FTDI chip (in the cable) could likely become overloaded, and go into shut down.

If you are using 5V to power the board, please make sure the supply can provide enough current for your LEDs as well.

ld21, 2016-06-23:
I tried with a more reliable power source and used the FTDI-cable only for the serial connection (using GND TX und RX) – no difference, millis() keeps returning the same value. Could it have to do something with the 8MHz of the 328V?

ld21, 2016-08-21:
I replaced the Lilypad with an arduino pro micro clone that uses a 32u4 – now everythings works perfectly and as expected. Thanks for your library! Dani

closed 2016-08-21 by ld21