bhagman/SoftPWM #4: SoftPWM_Wiring_HeartBeat

klint-k, 2017-11-16:
Arduino: 1.8.5 (Linux), Board: "Arduino Nano, ATmega328P"

/home/pi/Arduino/libraries/SoftPWM/examples/SoftPWM_Wiring_HeartBeat/SoftPWM_Wiring_HeartBeat.pde: In function 'void setup()':
SoftPWM_Wiring_HeartBeat:8: error: 'WLED' was not declared in this scope
   SoftPWMSet(WLED, 0);
              ^
/home/pi/Arduino/libraries/SoftPWM/examples/SoftPWM_Wiring_HeartBeat/SoftPWM_Wiring_HeartBeat.pde: In function 'void loop()':
SoftPWM_Wiring_HeartBeat:18: error: 'WLED' was not declared in this scope
   SoftPWMSet(WLED, 255);
              ^
exit status 1
'WLED' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

DoiChris, 2018-06-05:
even if I set the PWM to 0 the LED doesn't go of entirely. Its always lighting a bit! what can be the reason? I used a sink configuration with the ATMEGA GPIO to GND.

bhagman, 2018-06-07:
Fixed with Paul's changes here: https://github.com/bhagman/SoftPWM/pull/3

bhagman, 2018-06-07:
@DoiChris -- please create a separate issue

closed 2018-06-07 by bhagman