bhagman/SoftPWM #5: LED doesn't go off entirely!

DoiChris, 2018-06-07:
even if I set the PWM to 255 the LED doesn't go off entirely. Its always lighting a bit! What can be the reason? I used a sink configuration with the ATMEGA GPIO to GND.
[image]

bhagman, 2018-06-10:
Ah yes. Glitch free PWM.

https://stackoverflow.com/questions/23853066/how-to-achieve-zero-duty-cycle-pwm-in-avr-without-glitches

It's a bit troubling, and there may be a way to have both true off and true on (i.e. solid signals with no pulses), however the current version of SoftPWM has a single pulse when the value is at 255, unfortunately.

So, to get around this problem, I introduced a way to switch the polarity for individual pins so that you can specify which channels you want to have "true zero".

In your case, since you are sinking to control your LEDs, you need to switch the polarity of SoftPWM pins to `SOFTPWM_INVERTED`.  Here's how you do that:

```cpp
void setup()
{
  SoftPWMBegin(SOFTPWM_INVERTED);
  ...
}
```

You can also do it for individual pins, but it needs to be done _after_ at least the first `SoftPWMSet()` for the pin.

```cpp
  ...
  SoftPWMSet(MyLEDPin, 0);
  SoftPWMSetPolarity(MyLEDPin, SOFTPWM_INVERTED);
  ...
```

FYI: the "default" polarity is `SOFTPWM_NORMAL`

This should solve your problem.

DoiChris, 2018-06-11:
Hi Brett, thank you very much! This workaround worked perfect!

bhagman, 2018-06-11:
No problem!

closed 2018-06-11 by bhagman